riscv-boom/riscv-boom #688: The error occurred during testing of boom core

HanyJie, 2024-04-23:
**Type of issue**: question

Why does my scalaVersion use the default setting 2.13.10, but when testing boom core, it reports the following error, claiming it cannot find the rocketchip_2.13-1.6.0.pom file?

[image]

yoyo115956, 2024-06-18:
Indeed, I've encountered the same problem today, and feel that the likely reason is that the installer for rocketchip 2.13 is no longer being offered.
See this installer site https://repo1.maven.org/maven2/edu/berkeley/cs/

yoyo115956, 2024-06-18:
Maybe you can try the method here too, it might work:
https://github.com/ucb-bar/constellation/issues/50

yoyo115956, 2024-06-18:
The detailed code is as follows: 
git clone https://github.com/chipsalliance/rocket-chip.git
cd rocket-chip
git checkout 4fbd2f238db36b2862319e94c2f96d63bd52c98b
git submodule update --init --recursive
sbt "publishLocal"
sbt "project cde; set publishArtifact := true; publishLocal"
sbt "project rocket-macros; set publishArtifact := true; publishLocal"
sbt "project hardfloat; set publishArtifact := true; publishLocal"

yoyo115956, 2024-07-02:
I'm using sbt version 1.10.0, but the rocketchip on the given branch is already limited to version 1.3.13 and it's all working fine in my situation. 
https://constellation.readthedocs.io/en/latest/Introduction/Installation.html#installing-rocketchip. The code details are also referenced as mentioned by the original author here.

``` shell
(base) liyongting@LAPTOP-MRFD2B0N:~/rocket-chip$ sbt --version
sbt version in this project: 1.3.13
sbt script version: 1.10.0
```

``` shell
(base) liyongting@LAPTOP-MRFD2B0N:~/rocket-chip$ sbt "publishLocal"
[info] welcome to sbt 1.3.13 (Private Build Java 1.8.0_402)
[info] loading settings for project rocket-chip-build from plugins.sbt ...
[info] loading project definition from /home/liyongting/rocket-chip/project
[info] loading settings for project rocketchip from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
Using addons:
[info] set current project to rocketchip (in build file:/home/liyongting/rocket-chip/)
[info] Wrote /home/liyongting/rocket-chip/target/scala-2.13/rocketchip_2.13-1.6.0.pom
[info] Main Scala API documentation to /home/liyongting/rocket-chip/target/scala-2.13/api...
[warn] /home/liyongting/rocket-chip/src/main/scala/devices/debug/Debug.scala:957:28: Auto-application to `()` is deprecated. Supply the empty argument list `()` explicitly to invoke method unary_~,

......
[info]  published ivy to /home/liyongting/.ivy2/local/edu.berkeley.cs/rocketchip_2.13/1.6.0/ivys/ivy.xml
[success] Total time: 210 s (03:30), completed Jul 2, 2024 4:22:00 PM
```

SeddonShen, 2024-07-08:
> I'm using sbt version 1.10.0, but the rocketchip on the given branch is already limited to version 1.3.13 and it's all working fine in my situation. [...]

OK, I will try again , thx!!