riscv-boom/riscv-boom #689: About the Load Queue design in Boom

zhaoxiahust, 2024-04-23:
**Type of issue**: question

In boom, entries in the Load Queue (LDQ) are allocated in the decode stage. I cannot understand this design from the CPU micro-architecture view. In my opinion, the entry can be allocated as late as when the load instruction is issued from the load issue queue. Why Boom implements such an algorithm? Any help?

Kind regards

jerryz123, 2024-04-23:
Its a bit more challenging to allocate at load-issue.

Load-issue is out-of-order. What happens if you issue a bunch of younger loads, fill up the LAQ, and then a older load is issued?  You can't deallocate the younger LAQ entry, since that address may need to be used to resolve memory ordering, or detect a failed store->load hazard?

zhaoxiahust, 2024-04-23:
Jerry, thanks for the kind reply. Issuing is out of order but renaming and dispatching is in-order, right? Why not do this in the dispatching stage?

Even if you want to allocate a load entry in the issue stage, it is also possible. What you need to do is adding an age id to classify the older instructions or younger instructions just as Xuantie C910 does (https://github.com/T-head-Semi/openc910).

I want to learn more about the architecture insight behind this design which looks a bit not that elegant.

jerryz123, 2024-04-24:
> Why not do this in the dispatching stage?

After briefly reviewing the code, I realize it is being done in the dispatch stage? https://github.com/riscv-boom/riscv-boom/blob/506c11c99ca9f6015d14fdc02c45c1e22983d06f/src/main/scala/v4/exu/core.scala#L747-L753

> Even if you want to allocate a load entry in the issue stage, it is also possible.

Adding a age-id allows for age-disambiguation, but you might still run into the issue where younger loads occupy all the slots, and you have a deadlock situation from the structural hazard when attempting to issue an older load.
There are still ways to get around this, but all the solutions impose additional complexity. Getting a super-scalar out-of-order LSU correct is, in my opinion, the most challenging part of the entire microarchitecture.

> I want to learn more about the architecture insight behind this design which looks a bit not that elegant.

Keep in mind this is a graduate research project with a fraction of the engineering effort that would go into any commercially viable implementation.  I'm glad you observed the limitations of the current implementation. LSU CAMs are very costly structures, so maximizing the utilization of those structures is critical to building large-instruction-window designs.

zhaoxiahust, 2024-04-24:
Thanks！
If it is done in the dispatch stage, the manual and the LSU architecture figure (https://docs.boom-core.org/en/latest/sections/load-store-unit.html) might need to be updated accordingly.

jerryz123, 2024-04-24:
Oh yes, the figures and docs are very out of date, unfortunately.