riscv-boom/riscv-boom #690: Dual boom core simulation hang

crsullivan13, 2024-05-08:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**

https://github.com/riscv-boom/riscv-boom/issues/659 seems related.c
https://groups.google.com/g/chipyard/c/rOhedyr7O-4 opened discussion on the Chipyard forms.
Simulating on FireSim version 1.17.1. The configuration I am simulating is the following:
```
class FsimDualBoom2BankConfig extends Config(
  new freechips.rocketchip.subsystem.WithNBanks(2) ++
  new freechips.rocketchip.subsystem.WithInclusiveCache(nWays=16, capacityKB=1024) ++
  new freechips.rocketchip.subsystem.WithExtMemSize((1 << 30) * 4L) ++
  new WithDefaultFireSimBridges ++
  new WithDefaultMemModel ++
  new WithFireSimConfigTweaks ++
  new chipyard.DualLargeBoomConfig)
  
 class DualLargeBoomConfig extends Config(
  new chipyard.config.WithNPerfCounters(6) ++
  new boom.common.WithNLargeBooms(2) ++                          // 2 boom cores
  new chipyard.config.WithSystemBusWidth(128) ++
  new chipyard.config.AbstractConfig)
```
This issue has fired the pipeline hang assertion in Boom.

To reproduce, simulate the above design on FireSim. Run either two Disparity workloads from SD-VBS with the image input size of CIF or run two simple sequential write tasks that are LLC fitting (https://github.com/CSL-KU/IsolBench/blob/master/bench/bandwidth.c).

I should note that other pairings of SD-VBS workloads (which have lower LLC bandwidth than Disparity) do not experience the hang.

**What is the current behavior?**
Currently, when either of these workload pairs are run, the simulation hangs indefinitely with the pipeline hang assertion firing if synthesizable asserts are enabled.

**What is the expected behavior?**
These seem like normal workloads to want to simulate. The expected behavior is that the simulation would not hang.

**Please tell us about your environment:**

Boom commit hash 1a153d4.
Chipyard commit hash a88d69b.
os: Linux buildroot 6.2.0-00004-ge3e5d275a34f-dirty #25 SMP Tue May  7 22:04:17 CDT 2024 riscv64 GNU/Linux

**What is the use case for changing the behavior?**
These seem like perfectly reasonable workloads to run in a simulation environment without any issue.

jerryz123, 2024-05-08:
You are running the workload under Linux? Can you also try using v3 MegaBoom? That config should be the most robust

crsullivan13, 2024-05-08:
Yes, we are using buildroot Linux. I can give the MegaBoom a shot. By v3 MegaBoom do you mean something like the following (with two cores of course):
`class MegaBoomConfig extends Config(
  new boom.common.WithNMegaBooms(1) ++                           // mega boom config
  new chipyard.config.WithSystemBusWidth(128) ++
  new chipyard.config.AbstractConfig)`
  
Do you know if a dual core MegaBoom will fit on the VCU118? I have had issues trying to synthesize a quad LargeBoom setup.

jerryz123, 2024-05-08:
Yes, that config.

Dual core megaboom should work, from what I recall

crsullivan13, 2024-05-08:
I do have one more question while waiting for the design to synthesize. We are relying on the SiFive LLC in order to have a multi banked cache. Could the cache be part of the problem? If it is, is there a different way to have a multi-banked LLC in the simulation?

jerryz123, 2024-05-08:
Unlikely, but not impossible. Unfortunately, there is no alternative.

crsullivan13, 2024-05-09:
The design I synthesized with two MegaBooms is failing when trying to simulate (my other designs still simulate fine). The simulation gets stuck at "commencing simulation". I'll continue working on this and update once I have a working simulation.

jerryz123, 2024-05-09:
It is possible two megabooms do not fit

crsullivan13, 2024-05-10:
So here is what I have gotten. Dual MegaBooms can't boot Linux for some reason, there is a kernel panic. I ran a bare metal workload that I used to observe the hang in the past and the simulation does still seem to hang as before.

crsullivan13, 2024-05-28:
For confirmation I tested with two largebooms without the SiFive LLC in the design and still produced the pipeline hang. This happens for both running two sequential write tasks and when running two Disparity workloads from SD-VBS. 

module: BoomCore_1, path: FireSim.chiptop0.system.tile_prci_domain_1.tile_reset_domain.boom_tile.core]
Assertion failed: Pipeline has hung.
 at core.scala:1318 assert (!(idle_cycles.value(13)), \"Pipeline has hung.\")

crsullivan13, 2024-09-23:
I need to amend my previous comment. If I replace the SiFive LLC with the BroadcastManager, I am unable to reproduce the hang. I plan to test with the FASED timing model and with a single bank LLC as well just to see.

Additionally, I produced a waveform of a hanging metasimulation with a dual core LargeBoom with a 2 Bank 1MB LLC, each running a sequential write task that is L2 fitting (they don't share data and each access is a new cache line). 

From what I can see, eventually, each core's write-back buffers fill up and after actually sending the write back (should be a release on channel C I suppose), the write-back units never receive the expected ack response (Channel D) with the correct source bits. This hangs the entire pipeline.

 This explains why the hang only occurs with large amounts of write misses to the L1 and not with reads or with L1 fitting workloads. Each core experiences this, but one seems to have it happen earlier than the other, based on the waveform.

I am still working my way down to see why this might be happening. I suppose it must have something to do with SinkC, SourceD or maybe also the MSHRs in the LLC. @jerryz123 do you have any other suggestions based on what I have found so far?

jerryz123, 2024-09-23:
If you have a waveform of a hanging metasimulation, can you send it to me? I can take a look.

crsullivan13, 2024-09-24:
I did a bit more digging into the LLC. Found that the buffer in SinkC ends up filling up and never emptying out. Seems like SourceD eventually never gets the correct TL opcode/priority value (related I think) and doesn't pop off of SinkC buffer, as well as never sending the ack response. I assume there is an MSHR involved here, but still trying to understand the cache.

Also, I sent the waveform in a dropbox link to your email. Hopefully you got it.

jerryz123, 2024-09-26:
Hi Connor, can you clarify which configuration of boom (or commit hash) you were using?

crsullivan13, 2024-09-27:
To collect this waveform, I made sure to use a fresh FireSim repo with the hash b44c47149fa4dd7a3c6f5ff9be2cc5f920554f81.

The boom hash is 6a3ad0a1d9ae659aba5aea78978ef1a7cb5dd163.

I am using LargeBoom config from the v3 boom. Pasted exact configs below.

class LargeBoomDualConfig extends Config(
 new boom.v3.common.WithNLargeBooms(2) ++            
 new chipyard.config.WithSystemBusWidth(128) ++
 new chipyard.config.AbstractConfig)

class DualLargeBoomConfig extends Config(
 new freechips.rocketchip.subsystem.WithNBanks(2) ++
 new freechips.rocketchip.subsystem.WithInclusiveCache(nWays=16, capacityKB=1024) ++
 new WithDefaultFireSimBridges ++
 new WithDefaultMemModel ++
 new WithFireSimConfigTweaks ++
 new chipyard.LargeBoomDualConfig)