riscv-boom/riscv-boom #698: Load address misaligned exception can cause the meltdown attack on Boom v3.

Cpider, 2024-07-28:
**Type of issue**: bug report

**Impact**: rtl refactoring

**Development Phase**: proposal

**Other information**
Hi,
I found load address misaligned exception is another type of exception that can cause Meltdown-type attack on boom v3.
When **memaddrcalc** gets the unaligned address, it does not stop accessing it. And **LoadGen** will calculate incorrect data from the original aligned address data based on the misaligned address, and write them to the register file.
If the instruction that triggers the load address misaligned exception access the high-privilege address at the same time, such as trigger a load access exception or page fault exception. Boom will handle the misaligned exception first and leak part of the secret value. For example, the secret value 0xdeadbee0 is stored in 0x80800000, and when accessing 0x80800001, 0xdeadbebe will be obtained.
The attached PoC demonstrates a Meltdown attack that exploits the vulnerability to leak a portion of the secret from high-privilege software running in M-mode to low-privilege software running in S-mode.
[misaligned-meltdown.zip](https://github.com/user-attachments/files/16403026/misaligned-meltdown.zip)

This can be mitigated by fixing the Meltdown-type vulnerability.

tojauch, 2024-08-02:
This vulnerability has already been detected with the help of a formal tool called [UPEC](https://github.com/RPTU-EIS/upec-boom-verification-suite) in May 2021 and communicated with the BOOM developer team back then. 
They started to implement a fix (which UPEC still identified to be partly vulnerable) and then told us that they postpone this issue to BOOMv4.
The vulnerability has also been presented at [RISC-V Summit Europe 2023](https://riscv-europe.org/summit/2023/media/proceedings/posters/2023-06-06-Tobias-JAUCH-abstract.pdf) and was published in a paper presenting a [TES-secure version of BOOMv3](https://github.com/RPTU-EIS/SecureBOOM) at [ICCAD 2023](https://ieeexplore.ieee.org/abstract/document/10323843)

jerryz123, 2024-08-02:
Thanks, this is all very cool to see. It is a real-life demonstration of how microarchitectural design decisions easily inadvertently introduce side-channels.