riscv-boom/riscv-boom #704: Managing CSRs in BOOM

HakamAtassi, 2024-09-15:
Hello!

I was wondering if I could get some clarification on how BOOM manages CSRs, given that CSR instructions seem to go into the same RS as INT operations. Specifically:

1) How does the scheduler re-order consecutive writes to the same CSR when the CSR writes are ready for issue? None of the issue unit code seem to be exposed to CSRs. 

2) For CSR reads, the spec mentions the following: 

```
Each RISC-V hart normally observes its own CSR accesses, including its implicit CSR accesses, as
performed in program order. In particular, unless specified otherwise, a CSR access is performed after
the execution of any prior instructions in program order whose behavior modifies or is modified by
the CSR state and before the execution of any subsequent instructions in program order whose
behavior modifies or is modified by the CSR state. Furthermore, an explicit CSR read returns the CSR
state before the execution of the instruction, while an explicit CSR write suppresses and overrides any
implicit writes or modifications to the same CSR by the same instruction
```

Which has caused me to wonder, for reading something like `instret`, you obviously cant perform the read until all previous instructions have retired, while reading something like `cycle` can, in theory, be done as soon as its ready, but this can obviously lead to situations where the `cycle` reads will be fired off before whatever thing you are trying to time has executed because there is no real dependency here to resolve. How do you manage when these more time/execution sensitive registers are read, particularly when there is no obvious dependence?

For 2, I assume the use of FENCE is necessary to get accurate timing info, but my follow-up would be to ask where/how fences are managed in BOOM, because I also cant seem to find that information exposed in issue-unit.scala or similar. 

In general, I feel like I'm missing something because placing CSR operations in the same RS as Int operations feels awkward to me due to how much it complicates issuing instructions. With that said, I feel like this is mostly in my head. Please let me know if you can help clarify things or point me in the right direction. 

Thanks a lot for taking the time. Take care.

ccelio, 2024-09-15:
https://github.com/riscv-boom/riscv-boom/blob/d2a64f7ca9fd914d9c686cb23edcd32d3465a02e/src/main/scala/v4/exu/decode.scala#L153

CSR instructions are marked "unique" -- so if you're the only instruction in the issue queue and you're the head of the ROB, there's no extra complexity needed.

Of course, there's room for better performance here, especially if for example, it's doing something like reading a performance counter and you don't want to disturb the pipeline to heavily to do so...

HakamAtassi, 2024-09-15:
Ah okay that makes a lot of sense. Thank you very much :)