riscv-boom/riscv-boom #715: [BUG REPORT] Meltdown-type leakage through the TLB

TobiasKovats, 2025-01-12:
**Type of issue**: bug report

**Impact**: rtl refactoring

**Development Phase**: request

**Other information**

**What is the current behavior?**

[image]

Hi, 

Our microarchitectural fuzzer found a speculative meltdown-type leak through the TLB. We provide the relevant code snippet below: 

```riscv
... 

remu    a2, s0, gp # perform some long-latency instructions

rem     a2, a2, tp      

divw    ra, gp, a2     

blt     gp,ra,GADGET # ra is not ready yet, misspeculate to GADGET

... 

GADGET: 

ld   	sp,-2048(t1) # load from privileged page 

ld      s1,0(sp) # perform dependent access 
```
 
Although the load from the privileged address triggers a (transient) page-fault, the data is loaded into the register file. The transient window is sufficiently long to perform a dependant load. Therefore, the privileged data leaks to the program counter through the TLB. 
 
According to our experiments, the transient window is also sufficiently long to apply a mask to the privileged data before executing the dependant load, allowing a TLBleed-like attack. 
 
We attached the signal traces when leaking supervisor data from user mode: 
The transient load from the supervisor page (at D0100098000) (1st frame) triggers a page fault (0xD) (2nd frame). The privileged data (0xDEADBEEF) (3rd frame) is nevertheless loaded into the register file (4th frame), and a dependant load is performed (last frame). 
 
 
**What is the expected behavior?**

The dependant load should be killed due to the previous page fault.

**Please tell us about your environment:**

We use BOOM from Chipyard commit 004297b6 and Modelsim 2022.3_1 for simulation. We provide the ELF which can be executed with the setup from [Cascade](https://github.com/cascade-artifacts-designs/cascade-chipyard) to generate the signal traces. In the ELF, the speculated branch is at 0x800287d0/0xffffffd0100287d0 and the gadget at 0x80028e00.
[program.elf.zip](https://github.com/user-attachments/files/18390427/program.elf.zip)

Phantom1003, 2025-01-14:
I believe this is an unknown bug. It was first reported by [SpecDoctor](https://github.com/compsec-snu/specdoctor/blob/master/PoC/tlb-attack/tlb-exploit.c).

tojauch, 2025-01-14:
The susceptibility of the LSU in SonicBOOM to this class of attacks has already been detected with the help of a formal tool called [UPEC](https://rptu-eis.github.io/UPEC/index.html) in May 2021 and communicated with the BOOM developer team back then. 
They started to implement a fix (a branch called "meltdown-fix" from back then should still exist) which UPEC again identified to be partly vulnerable. Afterwards they told us that this fix will be to postponed to BOOMv4.
The vulnerability has also been presented at [RISC-V Summit Europe 2023](https://riscv-europe.org/summit/2023/media/proceedings/posters/2023-06-06-Tobias-JAUCH-abstract.pdf) and was published in a paper presenting a [TES-secure version of BOOMv3](https://github.com/RPTU-EIS/SecureBOOM) at [ICCAD 2023](https://ieeexplore.ieee.org/abstract/document/10323843). 
Feel free to test our [SecureBOOM](https://github.com/RPTU-EIS/SecureBOOM) implementation. In our fix, we added measures to prevent faulting loads from returning data.

TobiasKovats, 2025-01-14:
Hi!
Thank you both for these insights!

> I believe this is a known bug. It was first reported by [SpecDoctor](https://github.com/compsec-snu/specdoctor/blob/master/PoC/tlb-attack/tlb-exploit.c).

The PoC code demonstrates that the TLB can be used as side-channel when the load is not triggering a page-fault, i.e. is not a case of Meltdown-type leakage.

> The susceptibility of the LSU in SonicBOOM to this class of attacks has already been detected with the help of a formal tool called [UPEC](https://rptu-eis.github.io/UPEC/index.html) in May 2021 and [...]

I will test the SecureBOOM implementation and let you know if I find anything, thanks!

TobiasKovats, 2025-01-29:
Hi!

After testing the SecureBOOM implementation, I can verify that it is indeed not vulnerable to this attack. The patch kills the page-faulting load, preventing sensitive data to be loaded into the RF.
However this meltdown-type leakage has neither been reported in [RISC-V Summit Europe 2023](https://riscv-europe.org/summit/2023/media/proceedings/posters/2023-06-06-Tobias-JAUCH-abstract.pdf) nor [ICCAD 2023](https://ieeexplore.ieee.org/abstract/document/10323843). The former discloses a meltdown-type vulnerability that relies on the data-cache as side-channel, the latter mentions additionally that the TLB can cause a side-channel, as confirmed by https://github.com/compsec-snu/specdoctor/blob/master/PoC/tlb-attack/tlb-exploit.c, however does not show it's applicability to meltdown-type attacks.

Thus, we believe that this is the first PoC to demonstrate that the TLB can be successfully exploited for cross-privilege meltdown type attacks on BOOM.