riscv-boom/riscv-boom #716: Stalling Unique Instructions Results in System Failure

zhiyaot, 2025-01-23:
**Context**
Our team is currently working on a security project leveraging the Boom V3 core. We use Boom V3 as the "protected core" and perform security processing alongside Boom for each "in-flight" instruction. To ensure proper synchronization, we gate instructions at the commit stage of the Boom Core in case the security processing lags behind.

Following a discussion on the Boom mailing list ([reference](https://groups.google.com/g/riscv-boom/c/r0X4K70eUxg?pli=1)), we modified the can_commit logic to insert a stalling signal. Specifically, we updated the can_commit condition from:

```scala
can_commit(w) := rob_val(rob_head) && !(rob_bsy(rob_head)) && !io.csr_stall
```
to:
```scala
can_commit(w) := rob_val(rob_head) && !(rob_bsy(rob_head)) && !io.csr_stall && !io.enforce_stop
```
This allows us to make the commit stage treat certain instructions as though they have not been executed.

We use the FPGA bitstream built via [this repository from Xilinx employees](https://github.com/eugene-tarassov/vivado-risc-v).

**Problem**
This approach has worked well for many instructions; we can even delay commit cycles and observe the corresponding effects (e.g., delayed UART output). However, we’ve encountered system failures for **_unique instructions_** if enforce_stop is asserted for an extended period. This appears to result in a load failure or similar behavior.

We have yet to identify the precise delay threshold that triggers these failures.

As a workaround, we’ve implemented clock gating for Boom Core during these unique instructions. With this change, we’ve been able to complete fairly complex tasks, such as booting Linux and running various user-space processes, without issue.

**Question**
We’re seeking guidance to understand the root cause of this issue:

- Are there internal timers or other mechanisms within Boom V3 that could trigger this kind of failure if instructions are delayed at the commit stage for too long?
- Could you point us to specific sections of the codebase that might be relevant for debugging this behavior?

Any advice or pointers for further investigation would be greatly appreciated. We also have an example design that specifically trigger this behavior on an Alveo U200 board that we are are more than happy to share, along side with any source files that we used to generate the Boom Core.

jerryz123, 2025-01-29:
No, it should be possible to delay commit indefinitely. 

If you provide more information on how the failure manifests, I could try to think if there is another issue.

zhiyaot, 2025-02-05:
We appreciate your response and find it reassuring to know this is achievable. Our team is currently developing a design that reliably replicates this issue on the Alveo U200 FPGA. We're focusing on implementing only essential modifications to ensure a deterministic reproduction of the problem. Once complete, we would be more than happy to share our findings and design with you.

We currently observe this on the console when we try to boot Linux using OpenSBI as an early-stage boot loader.

[image]

The address (EPC) observed here corresponds to the first `csr` instruction of OpenSBI's trap handler.

```asm
00000000800003f8 <_trap_handler>:
    800003f8:	34021273          	csrrw	tp,mscratch,tp
    800003fc:	06523023          	sd	t0,96(tp) # 60 <_fw_start-0x7fffffa0>
    80000400:	300022f3          	csrr	t0,mstatus
    80000404:	00b2d293          	srli	t0,t0,0xb
    80000408:	0032f293          	andi	t0,t0,3
    8000040c:	0032a293          	slti	t0,t0,3
    80000410:	12fd                	addi	t0,t0,-1
```