riscv-boom/riscv-boom #722: [BUG REPORT] Load-Load Ordering Logic: Executed Field Condition Fails, Causing Excessive Load Latency

Heresyrac, 2025-02-22:
**Type of issue:** bug report

**Impact:** Performance 

**Development Phase:** proposal

**Other information**
**What is Current Behavior ?**
During performance analysis of Boom, we observed a significant performance drop when executing cyclic Load-Store-Load operations with the same address. The issue stems from the Load-Load ordering logic in Boom, where the executed field is not properly set, leading to unexpected load latency. 

Boom currently enforces ordering for Load operations with the same address.
Even when an older Load operation has successfully retrieved data(Forward from STQ), the system incorrectly assumes it is still pending.
This results in continuous emission of the kill_forward s1_kill signal, canceling wake-up requests for younger Loads.
Younger Loads repeatedly retry wake-up until the older Load completes commit and exits the LDQ

` 

     when (       do_ld_search(w)       &&
                   l_valid               &&
                   l_addr.valid          &&
                   !l_addr_is_virtual    &&
                   dword_addr_matches(w) &&
                   mask_overlap(w)) {

        val searcher_is_older = lcam_younger_load_mask(w)(i)
          // #Debug: Inspect why forward killed
        dontTouch(searcher_is_older)
        assert(IsOlder(lcam_ldq_idx(w), i.U, ldq_head) === searcher_is_older)
        when (searcher_is_older) {
          when ((l_executed || l_succeeded) &&
                !s1_executing_loads(i) && // If the load is proceeding in parallel we don't need to kill it
                l_observed) {        // Its only a ordering failure if the cache line was observed between the younger load and us
            //ldq_order_fail(i) := true.B
            // failed_load := true.B

	    // this case is no longer possible where an older search gets replayed and found that a younger search has already been released. Note that this is stronger than rvwmo.
            assert(false.B)
          }
        } .elsewhen (lcam_ldq_idx(w) =/= i.U) {
          // The load is older, and it wasn't executed
          // we need to kill ourselves, and prevent forwarding
	  // can also forward from the next cycle
          //#Debug: Inspect why forward killed

          when (!(l_executed && (l_succeeded || l_will_succeed))) { //### Here 
            s1_set_execute(lcam_ldq_idx(w))    := false.B
            when (RegNext(dmem_req_fire(w) && !s0_kills(w)) && !fired_load_agen(w)) {
              io.dmem.s1_kill(w)               := true.B
            }
            kill_forward(w)                    := true.B
            //#Debug: Inspect why forward killed
            conflict_src(w)                    := i.U
          }
        }
      }
    }
  }`
`

[image]

[image]
We found LDQ_4(older load) already succeed, load in LDQ_6(younger load) can forward from STQ_1D, but LDQ_4.executed=0 cause kill_forward.
40 cyc later, older load committed & deq from LDQ, younger load wakeup and get forward from STQ_1D.
**Root Cause:**

The executed field in the LDQ entry is not set to 1 when an older Load retrieves data through STQ forwarding.
This prevents the condition (l_executed && (l_succeeded || l_will_succeed)) from being satisfied.
**Performance Impact:**

This issue causes excessive latency (40-80 clock cycles).
Unfinished instructions pile up, filling the freelist in the Rename stage.
Significant performance degradation occurs as a result.
**Expected Behavior**
When an older Load successfully retrieves data, the ordering condition should be considered satisfied, and younger Loads should no longer be blocked.

Potential Solution
We suspect that the trigger condition for kill_forward should be modified to:
(l_executed || (l_succeeded || l_will_succeed))
Please confirm if this assumption is correct.

**Please tell us about your environment?**
Master branch on Chipyard
**What is the use case for changing the behavior?**
Scenario: Cyclic execution of Load-Store operations, where Load-Store may maintain the same address across iterations.
Issue Manifestation: Significant performance degradation in same-address Load-Store cycles.
**Additional Questions**
Is the current same-address Load ordering code solely intended to prevent Load-Load ordering failures?
Are there other potential impacts that need to be considered?

tianrui-wei, 2025-02-27:
Hi @Heresyrac,

Could you state the workload you're running for the performance drop?

Sadly I don't think `or`'ing the signal together would be correct.
It also seems like ldq_4 entry was forwarded?

Tianrui

tianrui-wei, 2025-03-04:
Hi @Heresyrac,

Thanks for the bug report and clarification. Is it possible to share a (minimal working) example of your kernel for me to run?

The or'ing would be incorrect for the following scenario:

```
lw  // miss/address not available
// external probe
lw // hit, returns immediately
```
The second load in this case will be violating the memory order.

If a load forwards from a write, then additional logic must be added so that all probes to that address must be blocked, which is quite possible to cause deadlocks in the system.

Thanks,
Tianrui

Heresyrac, 2025-03-05:
> Hi [@Heresyrac](https://github.com/Heresyrac),
>
> Thanks for the bug report and clarification. Is it possible to share a (minimal working) example of your kernel for me to run?
>
> The or'ing would be [...]

Hi @tianrui-wei ,
1. Sorry, I cannot provide an example because it used company confidential information. However, this scenario should be easy to reproduce. I believe that repeatedly performing store-load operations on the same address can reproduce the issue.
2. I don't understand why a store-load forward would need to block probes on the same address. I think this is unnecessary, even if an older load occurs after these probes, the sequence of load(older) success -> load(older) commit -> store commit -> store write to dcache is strictly ordered. Therefore, the older load will get the old data, and the younger load will get the young data, which should not violate RVWMO.

closed 2025-03-05 by tianrui-wei

Heresyrac, 2025-03-06:
> 你好[@Heresyrac](https://github.com/Heresyrac)，
> 
> 抱歉，上述代码序列指的是您提议的更改为`&&`。`||`在这种情况下，处理新提议的更改和 PnR 会很棘手。
> 
> ```
> lw  // lw1
> sw // lw2 forwards from this
> // other constructs
> lw // lw2
> ```
> 
> 假设它们都通过了 PnR，那么在 sw 之后直到 lw 为止，对该线路的探测都必须被阻止，否则如果探测被确认，则必须重放 lw2，从而破坏 PnR。
> 
> 我做了一些本地实验，我认为你指出的更改确实应该恢复。我还没有想到一个干净的解决方案，当我想到一个时我会恢复这个帖子。欢迎提出替代解决方案，很高兴收到你的来信 :)
> 
> 天瑞

Hi Tianrui,
```
      when (       do_ld_search(w)       &&
                   l_valid               &&
                   l_addr.valid          &&
                   !l_addr_is_virtual    &&
                   dword_addr_matches(w) &&
                   mask_overlap(w)) {

        val searcher_is_older = lcam_younger_load_mask(w)(i)
          // #Debug: Inspect why forward killed
        dontTouch(searcher_is_older)
        assert(IsOlder(lcam_ldq_idx(w), i.U, ldq_head) === searcher_is_older)
        when (searcher_is_older) {
          when ((l_executed || l_succeeded) &&
                !s1_executing_loads(i) && // If the load is proceeding in parallel we don't need to kill it
                l_observed) {        // Its only a ordering failure if the cache line was observed between the younger load and us
            //ldq_order_fail(i) := true.B
            // failed_load := true.B

	    // this case is no longer possible where an older search gets replayed and found that a younger search has already been released. Note that this is stronger than rvwmo.
            assert(false.B)   // <-Here?
          }
        } .elsewhen (lcam_ldq_idx(w) =/= i.U) {
```
Is this code branch being triggered, causing a failed_load (manifesting as an assert not being met?), which in turn leads to a breaking PnR?

1. I'm unclear why we're entering this condition. If lw2 is executing and lw1 has already succeed obtained a forward from the previous store, I believe lw1 shouldn't need to re-search the LDQ afterward.
Is lw1 in an executed state where the dcache request has been issued but resulted in a miss? And is the (succeed || will_succeed) condition not being met?
If that's the case, modifying !(l_executed || (l_succeeded || l_will_succeed)) to !(l_succeeded || l_will_succeed) might resolve the issue?

2. I believe that even if lw1 hasn't succeeded by the time lw2 executes, failed_load shouldn't be triggered in this scenario. There should be no need to replay lw2 or block the probe because the presence of a store between loads shouldn't violate RVWMO. I think adjusting the assert condition might help:
```

((l_executed || l_succeeded) &&
 !s1_executing_loads(i) && // If the load is proceeding in parallel, we don't need to kill it
 l_observed &&
 !(l_forward_std_val && isOlder(lcam_uop.stq_idx, l_forward_stq_idx, stq_head)))
// Maybe? I'm not entirely sure if comparing lw1's uop.stq_idx with forward_stq_idx can accurately determine program order between lw1 and sw, but it might be theoretically possible.
```

2. To further improve performance, I think modifying the kill_forward logic could help. If the store providing the forward is younger than the load causing kill_forward, then kill_forward shouldn't be triggered. This should enhance load reordering capability by avoiding unnecessary kill_forwards. 
However, in my benchmark, I haven't observed this scenario yet (or perhaps I made some mistakes), so the OoO Load didn't result in a performance improvement.