riscv-boom/riscv-boom #726: Does BOOM support RoCCCSRs?

ZouYuyang, 2025-03-02:
LazyRoCC provides interfaces to add RoCC custom CSRs in rocket. However, changing rocket to BOOM resulting in firtool error says CustomCSRIO's cpu side signal is not connected. I've checkout BOOMTile implmentation and found out they are completely unconnected.

Is this a bug? or it is not supported in plan?

Thanks

jerryz123, 2025-03-02:
Are you looking at v3 or v4?

ZouYuyang, 2025-03-03:
I tried both, but none of it works. This is what i got

```
error: sink "rocc.io_csrs_0_ren" not fully initialized in "BoomTile"
error: sink "rocc.io_csrs_0_wen" not fully initialized in "BoomTile"
error: sink "rocc.io_csrs_0_wdata" not fully initialized in "BoomTile"
error: sink "rocc.io_csrs_0_value" not fully initialized in "BoomTile"
```

My configuration is

```
class myConfig extends Config(
  new xxx ++ // my rocc acc
  new boom.v3.common.WithNSmallBooms(1) ++
  new chipyard.config.AbstractConfig
)
```

Thanks